gmag11/ESPNtpClient #3: [pio] Error: Nothing to build. Please put your source code files

hpsaturn, 2020-11-30:
I think that we have a issue with `pio run` on root directory:

after clone, when I run it, I have on all environments:

```python
Error: Nothing to build. Please put your source code files to '/home/avp/tmp/src/ESPNtpClient/examples' folder
========================================================= [FAILED] Took 0.26 seconds =========================================================

Environment            Status    Duration
---------------------  --------  ------------
advanced_esp32         FAILED    00:00:00.252
advanced_esp8266       FAILED    00:00:00.323
esp32_basic_example    FAILED    00:00:00.252
esp8266_basic_example  FAILED    00:00:00.256
esp32_led_flasher      FAILED    00:00:00.250
esp8266_led_flasher    FAILED    00:00:00.257
==================================================== 6 failed, 0 succeeded in 00:00:01.591 ====================================================
```

hpsaturn, 2020-11-30:
I tryed to replicate it in my library, and I have the same issue on `*.ino` projects, but in my `platformio` example, works fine:

https://github.com/kike-canaries/canairio_sensorlib/blob/av/ci_improvement/platformio.ini

gmag11, 2020-11-30:
I do not recommend building that from root folder. Better, import an example as an Arduino project and build it individually. That plaformio.ini on root is only used for development testing

hpsaturn, 2020-12-01:
I'm just using platformio, and all works fine, but I was try to add a workflow on your repo, because it has not CI, for this reason I think that issue should be to resolve. The idea be a send a pull request with that. Let me review the issue..

gmag11, 2020-12-01:
In order to compile all variants from root you need to rename ino files to cpp. I guess it is a PlatformIO bug.

hpsaturn, 2020-12-01:
I added a PR for that the proyect will have a continous integration with github workflow, for now it only will be test the two architectures. But maybe a better clean solution is fix the *.ino files issue.

hpsaturn, 2020-12-06:
No problem, but we could leave open this issue because the PR only give CI compiling test feature to this library, but this issue exists yet.

gmag11, 2020-12-08:
Sure!