gmag11/ESPNtpClient #4: NTP.getTime not public

RaffiCologne, 2020-12-01:
Good morning, using your Lib as replacement for NTPClientLib.
Everything is running fine till now.
Using NTP.getTime() results in 
C:\Users\raffi\Documents\Arduino\libraries\ESPNtpClient-main\src/ESPNtpClient.h:153:10: error: 'void NTPClient::getTime()' is protected

     void getTime ();

Looking in ESPNtpLib.h you commented:

 /**
    * Starts a NTP time request to server. Returns a time in UNIX time format. Normally only called from library.
    * Kept in public section to allow direct NTP request.
    */
    void getTime ();

Is there another function to manually fire a request or what i am doing wrong?
Regards,
Ralf

gmag11, 2020-12-01:
Just fixed in last release. Why do you need that? I'd like to know the use case.

RaffiCologne, 2020-12-02:
Hi, thanks for your fast reply.

The picture shows a part of a web-interface.

If the user clicks the Button, time sync should happen manually.

gmag11, 2020-12-03:
Thanks, then I close this issue

closed 2020-12-03 by gmag11