gmag11/ESPNtpClient #6: Basic example doesn't compile

theotherjenkutler, 2020-12-03:
OSX 10.14.6
Arduino IDE 1.8.13

While compiling the basic example for Sparkfun ESP32 Thing Plus I have this error:
/Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/src/ESPNtpClient.cpp: In member function 'void NTPClient::getTime()':
/Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/src/ESPNtpClient.cpp:502:28: error: ambiguous overload for 'operator==' (operand types are 'IPAddress' and 'u32_t {aka unsigned int}')
     if (ntpServerIPAddress == INADDR_NONE) {
                            ^
/Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/src/ESPNtpClient.cpp:502:28: note: candidate: operator==(uint32_t {aka unsigned int}, u32_t {aka unsigned int}) <built-in>
In file included from /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc2/cores/esp32/Arduino.h:150:0,
                 from /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/src/ESPNtpClient.h:13,
                 from /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/src/ESPNtpClient.cpp:1:
/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc2/cores/esp32/IPAddress.h:63:10: note: candidate: bool IPAddress::operator==(const IPAddress&) const
     bool operator==(const IPAddress& addr) const
          ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc2/libraries/WiFi
 Not used: /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/libraries/WiFi

gmag11, 2020-12-03:
I would say that your ESP32 environment is not updated to last version. Try to update it using boards manager.

theotherjenkutler, 2020-12-03:
Thanks so much for your reply. ESP32 board is the latest version - 1.0.5-rc3
I just updated from 1.0.5-rc2 and it doesn't seem to have solved the problem. I have the same error.

theotherjenkutler, 2020-12-03:
Just in case its useful here is the entire verbose output:

/Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/hardware -hardware /Users/JenKutler/Library/Arduino15/packages -hardware /Users/JenKutler/Documents/Arduino/hardware -tools /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/tools-builder -tools /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/JenKutler/Library/Arduino15/packages -built-in-libraries /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/libraries -libraries /Users/JenKutler/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_build_446618 -warnings=none -build-cache /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_cache_780596 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -verbose /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/examples/basicExample/basicExample.ino
/Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/hardware -hardware /Users/JenKutler/Library/Arduino15/packages -hardware /Users/JenKutler/Documents/Arduino/hardware -tools /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/tools-builder -tools /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/JenKutler/Library/Arduino15/packages -built-in-libraries /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/libraries -libraries /Users/JenKutler/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_build_446618 -warnings=none -build-cache /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_cache_780596 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-96-g2852398-5.2.0.path=/Users/JenKutler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0 -verbose /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/examples/basicExample/basicExample.ino
Using board 'esp32' from platform in folder: /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3
Using core 'esp32' from platform in folder: /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3
bash -c "[ ! -f /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/examples/basicExample/partitions.csv ] || cp -f /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main/examples/basicExample/partitions.csv /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_build_446618/partitions.csv"
bash -c "[ -f /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_build_446618/partitions.csv ] || cp /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/partitions/default.csv /var/folders/vt/hk6mczq929bbmxtmdc6mbxc80000gn/T/arduino_build_446618/partitions.csv"
Detecting libraries used...
/Users/JenKutler/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-96-g2852398-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/sdk/include/config -I/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/sdk/include/app_trace -I/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/sdk/include/app_update -I/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/sdk/include/asio -I/Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/tools/sdk/include/bootloader_support -I/Users/JenK [... cut ...]
 Not used: /Users/JenKutler/Downloads/Arduino 2.app/Contents/Java/libraries/WiFi
Using library ESPNtpClient-main at version 0.2.0 in folder: /Users/JenKutler/Documents/Arduino/libraries/ESPNtpClient-main 
Using library WiFi at version 1.0 in folder: /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/libraries/WiFi 
Using library Ticker at version 1.1 in folder: /Users/JenKutler/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc3/libraries/Ticker 
exit status 1
Error compiling for board ESP32 Dev Module.

gmag11, 2020-12-05:
This should be fixed in https://github.com/gmag11/ESPNtpClient/tree/dev

closed 2020-12-12 by gmag11