gmag11/ESPNtpClient #12: fatal error: TZ.h: No such file or directory

kendo55, 2020-12-20:
Hi
and thanks for your great work!
With ESP32 it works well!
With ESP8266, there is a problem with in 
ESPNtpClient.h
```
#ifdef ESP32
#include "TZdef.h"
#else
#include "TZ.h"
#endif
```
there is no TZ.h !!!!

gmag11, 2020-12-20:
You need to update your esp8266 core as TZ.h is included there: https://github.com/esp8266/Arduino/blob/8375faa5427f70ad7542be3969f99c12cc11f015/cores/esp8266/TZ.h

kendo55, 2020-12-20:
[image]
I have installed the newest Version 2.7.4. in the Bordmanager!
and there is still missing the TZ.h
Thanks!

kendo55, 2020-12-20:
[image]
The TZ.h file is in the directory:
D:\arduino_1_8_5_ESP8266_r2_7_4\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266

kendo55, 2020-12-20:
I copied TZ.h file in your Lib-path.

Could you please compile your "advancedExample" in the Arduino-IDE and ESP8266 newest Version 2.7.4. Where are compiling error like this:

```

NTP_Client_HS:37: error: ntpServer causes a section type conflict with __c

 const PROGMEM char* ntpServer = "pool.ntp.org";

                     ^

In file included from D:\arduino_1_8_5_ESP8266_r2_7_4\hardware\esp8266com\esp8266\cores\esp8266/Arduino.h:241:0,

                 from D:\projekte\C_Control\ESP8266\sketchbook\NTP_Client_HS\NTP_Client_HS.ino:9:

D:\arduino_1_8_5_ESP8266_r2_7_4\hardware\esp8266com\esp8266\cores\esp8266/pgmspace.h:16:51: note: '__c' was declared here

 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))

                                                   ^
```
My "version" : const char ntpServer[] PROGMEM = "pool.ntp.org";   works....

But the next error was:
```
D:\projekte\C_Control\ESP8266\sketchbook\libraries\ESPNtpClient-main\src\ESPNtpClient.cpp: In member function 'void NTPClient::getTime()':

D:\projekte\C_Control\ESP8266\sketchbook\libraries\ESPNtpClient-main\src\ESPNtpClient.cpp:533:58: error: cannot convert 'ip_addr*' to 'const ip_addr_t* {aka const ip4_addr*}' for argument '2' to 'err_t udp_connect(udp_pcb*, const ip_addr_t*, u16_t)'

     result = udp_connect (udp, &ntpAddr, DEFAULT_NTP_PORT);
```

kendo55, 2020-12-21:
sorry for disturb you! 
It was a Update problem.
In a new Installation with Version 2.7.4. it works well!

closed 2020-12-21 by kendo55