gmag11/ESPNtpClient #17: ESPNtpClient won't compile with TimeLib.h

morres83, 2021-03-12:
Hello, I am facing the following issue:
The ESPNtpClient doesn't compile on ESP32 as soon as the TimeLib.h (paulstoffregen/Time@^1.6) gets included. Unfortuneatly this library is a dependency of DS3232RTC.h which I also need. Without Time.h and TimeLib.h, there is no compiling error

Is there a workaround?

>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:65:16: error: expected ')' before '(' token
 constexpr auto SECS_PER_MIN = ((time_t)(60UL));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:66:16: error: expected ')' before '(' token
 constexpr auto SECS_PER_HOUR = ((time_t)(3600UL));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:67:16: error: expected ')' before '(' token
 constexpr auto SECS_PER_DAY = ((time_t)(SECS_PER_HOUR * 24UL));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:68:16: error: expected ')' before '(' token
 constexpr auto DAYS_PER_WEEK = ((time_t)(7UL));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:69:16: error: expected ')' before '(' token
 constexpr auto SECS_PER_WEEK = ((time_t)(SECS_PER_DAY * DAYS_PER_WEEK));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:70:16: error: expected ')' before '(' token
 constexpr auto SECS_PER_YEAR = ((time_t)(SECS_PER_DAY * 365UL));

                
>.pio/libdeps/nodemcu-32s/ESPNtpClient/src/ESPNtpClient.h:71:16: error: expected ')' before '(' token
 constexpr auto SECS_YR_2000 = ((time_t)(946684800UL)); ///< @brief The time at the start of

morres83, 2021-03-12:
Actually the workaround is to just comment out these lines or wrap in into a
#ifndef SECS_PER_MIN
...
#endif

These items are already defined in Time.h as it seems.

gmag11, 2021-03-14:
That's a good solution. Would you add a pull request?

gmag11, 2021-03-16:
Fixed in https://github.com/gmag11/ESPNtpClient/releases/tag/0.2.5

closed 2021-03-16 by gmag11