gmag11/ESPNtpClient #23: begin resetting the NTP server

abaskin, 2021-09-11:
The begin method sets the NTP server if it is called without a value. I find this behavior unintuitive and at minimum suggest that the behavior should be documented. To give an example. 

NTP.setNtpServerName("asia.pool.ntp.org");
NTP.begin();

In the code above the NTP server is reset to the default by begin even though it was previously set by setNtpServerName. A better approach would be for begin to only set the NTP server value to the default is it has not already been set.

gmag11, 2021-10-04:
Hello. You are right. I'll make that a prior call to `NTP.setNtpServerName(...)` force begin to use assigned server if it is called without arguments. Thank you

gmag11, 2021-10-04:
This should be working as expected after 8be062413c3a410a4677c6e7148fd1b2aa90cd5a, on **dev** branch. @abaskin Please test it and give some feedback

abaskin, 2021-10-05:
I changed my code to set the NTP server before calling begin, pointed to the dev branch and it workes perfectly. Thank you.

closed 2021-10-05 by gmag11