gmag11/ESPNtpClient #24: build problems - 'class Ticker' has no member named 'detach'

tiffoknee, 2021-11-02:
I'm trying to replace ntpclient with espntpclient, and also update the platform from 2.6.0 to 3.2.0 and I'm getting these errors when I try to build:

In file included from include/main.h:20,
                 from src/main.cpp:1:
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h: In member function 'void NTPClient::stop()':
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h:398:19: error: 'class Ticker' has no member named 'detach'
  398 |         loopTimer.detach ();
      |                   ^~~~~~
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h:399:23: error: 'class Ticker' has no member named 'detach'
  399 |         receiverTimer.detach ();
      |                       ^~~~~~
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h:401:23: error: 'class Ticker' has no member named 'detach'
  401 |         responseTimer.detach ();
      |                       ^~~~~~
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h: In member function 'char* NTPClient::getTimeStr(timeval)':
.pio/libdeps/d1_mini/ESPNtpClient/src/ESPNtpClient.h:557:35: error: cannot convert 'suseconds_t*' {aka 'long int*'} to 'const time_t*' {aka 'const long long int*'}
  557 |         tm* local_tm = localtime (&moment.tv_usec);
      |                                   ^~~~~~~~~~~~~~~

tiffoknee, 2021-11-02:
actually, the problem is higher up.. no matching function for call to 'Ticker::Ticker()' in the NTPClient class

gmag11, 2021-11-22:
New library does not use Ticker but internal timers, Maybe you still have the old library files in `.pio` directory.

closed 2021-12-17 by gmag11