gmag11/ESPNtpClient #25: error: 'timezone' does not name a type

tatuiot, 2021-11-10:
Hi , thanks for your library. 
I managed to make it work commenting out the line 
timezone timeZone;              ///< @brief
But it happens that the time displayed is 6hrs ahead of the real time... It was working last week... but now... suddenly... it stopped working properly and I cannot make it work again.

I am using VSCode 1.62.1 / Platformio 5.2.3 / ESP32 Dev Kit1 / ESPNtpClient 0.2.5

```
uint8_t initNTP(void)
{
    NTP.setTimeZone(TZ_America_Argentina_Buenos_Aires);
    NTP.setInterval(600);
    NTP.setNTPTimeout(NTP_TIMEOUT);
    //NTP.setMinSyncAccuracy (5000);
    //NTP.settimeSyncThreshold (3000);
    NTP.begin(ntpServer);
    Serial.println(F("NTP CONFIGURED"));
    return 0;
}
```
I am lost... thanks in advance any tip...

tatuiot, 2021-11-11:
Hi, I managed to fix it using the following statement
NTP.setTimeZone(TZ_Etc_GMTm3); 
there must be some kind of bug there because Buenos Aires time adds 3 hrs to GMT time and it should be -3.

gmag11, 2021-11-22:
Encoding for Buenos Aires time is `#define TZ_America_Argentina_Buenos_Aires	PSTR("<-03>3")`. I've found no reference for this format but https://developer.ibm.com/articles/au-aix-posix/ Notation using `<>` is not described there so maybe it is a mistake.

As Argentina does not have DST you can use GMT-3.

gmag11, 2021-12-25:
I've added an issue in ESP8266 Arduino core repository: https://github.com/esp8266/Arduino/issues/8423