gmag11/ESPNtpClient #28: ESP8266WiFiMulti.h

stoffel2, 2021-11-30:
Hi,

Thanks for the really great library which I use to synchronize multiple devices. I've had to switch to the dev branch and have not gotten it to work with ESP8266WiFiMulti.h. Perhaps the connect events are not being seen.

Best Regards,

Chris

gmag11, 2021-12-17:
Hi. Maybe you can try to use basicExample code: https://github.com/gmag11/ESPNtpClient/blob/main/examples/basicExample
As this does not use events to start sync you may get wrong time until device gets connected to the Internet.

stoffel2, 2021-12-17:
Hi, I've gotten WiFiMulti to work in the dev branch using ideas from the advancedExample. I discovered that putting any code such as a delay(1000) after NTP.begin(ntpServer) in my void setup reliably caused the a seg fault (for whatever reason), which I had mistakenly attributed to MiFiMulti. Regarding connectivity: I make sure my device gets synced using your event handler, which is very convenient.

Thanks again!

Chris

gmag11, 2021-12-18:
As I understand you solved your problem I close this issue. Feel free to reopen it if needed. Thanks.

closed 2021-12-18 by gmag11