gmag11/ESPNtpClient #33: error: cannot convert 'suseconds_t*' {aka 'long int*'} to 'const time_t*' {aka 'const long long int*'}

Schawen, 2021-12-24:
I tried to compile an older project and experienced several problems with NTPClientLib.
I now replaced with the recommended ESPNtpClient library but get the following errors:

`In file included from .pio\libdeps\d1_mini_pro\ESPNtpClient\src\ESPNtpClient.cpp:1:
.pio\libdeps\d1_mini_pro\ESPNtpClient\src\ESPNtpClient.h: In member function 'char* NTPClient::getTimeStr(timeval)':
.pio\libdeps\d1_mini_pro\ESPNtpClient\src\ESPNtpClient.h:557:35: error: cannot convert 'suseconds_t*' {aka 'long int*'} to 'const time_t*' {aka 'const long long int*'}
  557 |         tm* local_tm = localtime (&moment.tv_usec);
      |                                   ^~~~~~~~~~~~~~~
      |                                   |
      |                                   suseconds_t* {aka long int*}
In file included from C:\Users\MiKo\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/HardwareSerial.h:31,
                 from C:\Users\MiKo\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:288,
                 from .pio\libdeps\d1_mini_pro\ESPNtpClient\src\ESPNtpClient.h:13,
                 from .pio\libdeps\d1_mini_pro\ESPNtpClient\src\ESPNtpClient.cpp:1:
c:\users\miko\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\time.h:64:37: note:   initializing argument 1 of 'tm* localtime(const time_t*)'
   64 | struct tm *localtime (const time_t *_timer);
      |                       ~~~~~~~~~~~~~~^~~~~~
*** [.pio\build\d1_mini_pro\lib242\ESPNtpClient\ESPNtpClient.cpp.o] Error 1
In file included from src\main.cpp:17:
.pio\libdeps\d1_mini_pro\ESPNtpClient\src/ESPNtpClient.h: In member function 'char* NTPClient::getTimeStr(timeval)':
.pio\libdeps\d1_mini_pro\ESPNtpClient\src/ESPNtpClient.h:557:35: error: cannot convert 'suseconds_t*' {aka 'long int*'} to 'const time_t*' {aka 'const long long int*'}
  557 |         tm* local_tm = localtime (&moment.tv_usec);
      |                                   ^~~~~~~~~~~~~~~
      |                                   |
      |                                   suseconds_t* {aka long int*}`

Where is the Problem?

gmag11, 2021-12-25:
Hello. This information is not enough to diagnose your issue. Please add more information like: platform you are developing in (ESP32, ESP8266), Arduino core that you are using, version of this library that you have included.

If possible, add a link to your code so that we can try to reproduce this error.

Schawen, 2021-12-26:
Sorry, here are the missing informations (also added to the top entry) :

PLATFORM: Espressif 8266 (3.2.0) > WeMos D1 mini Pro
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash       
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2) 
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)        

Library version:
gmag11/ESPNtpClient@^0.2.5

As a workaround - to get the project compiled -  I changed line 557 from
`tm* local_tm = localtime (&moment.tv_usec);
`
to
`tm* local_tm = localtime (&moment.tv_sec);
`
in function:
`    char* getTimeStr (timeval moment) {
        //tm* local_tm = localtime (&moment.tv_usec);
        tm* local_tm = localtime (&moment.tv_sec);
        size_t index = strftime (strBuffer, sizeof (strBuffer), "%H:%M:%S", local_tm);
        snprintf (strBuffer + index, sizeof (strBuffer) - index, ".%06ld", moment.tv_usec);
        return strBuffer;
    }
`
where **tv_usec** and **tv_sec** are defined in 
`struct timeval {
	time_t		tv_sec;		/* seconds */
	suseconds_t	tv_usec;	/* and microseconds */
`

gmag11, 2021-12-29:
Hello,

I've reformatted your messages and realized that it was a bug. Your solution is correct. I've updated it on dev branch, in https://github.com/gmag11/ESPNtpClient/commit/06a321cead8e2a0cf163b67c7f855fc1e3020be3

Thank you!

closed 2021-12-29 by gmag11