gmag11/ESPNtpClient #37: ESPNtpClient.cpp: 483:50: error: 'struct pbuf' has no member named 'type

unregistered33, 2022-01-17:
First, thanks for excellent library!  But i have litle problem/
When i use esp32 core 1.0.6 the library work fine, but when i try to use core 2.0.2 i have error:
_ESPNtpClient.cpp: 483:50: error: 'struct pbuf' has no member named 'type

ESPNtpClient.cpp: 754:21: error: 'struct pbuf' has no member named 'type_

unregistered33, 2022-01-17:
looked at the pbuf file and did not find the field "type", but there is "type_internal". Im change it and code compile succesful. Is it right solution?

gmag11, 2022-01-22:
Hi, I'm using Platformio ESP32 framework. I guess it uses ESP32 Arduino core 1.0.6 so most probably you are right. I leave this open until PlatformIO supports newer core releases