gmag11/ESPNtpClient #38: ESP8266 reset

Laserlicht, 2022-02-06:
I get with the following code unfortunately after a few seconds always a reset at the NodeMCU. But the synchronization works, as you can see on the output. What is the problem?

Code:
```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESPNtpClient.h>

void setup() {
    Serial.begin(115200);

    WiFi.begin();
    while (WiFi.status() != WL_CONNECTED) {
        delay(100);
    }
    Serial.println(WiFi.localIP());

    NTP.setTimeZone(TZ_Etc_UTC);
    NTP.begin();
}

void loop() {
    String result = NTP.getTimeDateString(time(NULL), "%04Y-%02m-%02dT%02H:%02M:%02S,");
    uint64_t ms = NTP.millis() - (time(NULL) * 1000.0);
    char buffer[5];
    sprintf(buffer, "%03dZ", (int)ms);
    result += buffer;
    Serial.println(result);

    delay(100);
}
```

Platformio.ini:
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp01_1m]
platform = espressif8266
board = nodemcu
framework = arduino
lib_deps = 
	gmag11/ESPNtpClient@^0.2.6
	Ticker
monitor_speed = 115200
upload_speed = 115200
```

Output:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00048b60
~ld
192.168.2.123
[... 330 lines cut ...]
3ffffe60:  3ffee8a0 f2a78000 00000000 4020d2d8  
3ffffe70:  3fffff50 3fffff40 0000000c 3ffeeb8c  
3ffffe80:  00000000 00000020 00000020 401009ac  
3ffffe90:  00000003 00000001 3fffff50 401009e4  
3ffffea0:  e8b90478 0005d75a 000f4240 3ffeeb8c  
3ffffeb0:  cf97bc98 4277ecf9 7bcad000 40208f01  
3ffffec0:  3fffff74 f4240000 7ffffffb 3ffeeb8c  
3ffffed0:  3ffefd8c 00000020 3fffff50 4020301d  
3ffffee0:  00000000 00000002 3ffe89b0 402044a8  
3ffffef0:  3ffe89b1 00000018 3ffefdcb 402044a8  
3fffff00:  00000139 00000139 3ffe8644 401007a3  
3fffff10:  40202a28 3ffe89af 3ffeeb00 3ffeeb8c  
3fffff20:  40105ad5 001c719b 3ffeebf0 00000000  
3fffff30:  3ffee440 3ffeebf0 00000064 3ffeebf0  
3fffff40:  cf97bc98 3ffeeb8c 40203398 3fffefa0  
3fffff50:  3ffeebf0 cf97bc98 3ffeeb8c 40203a46  
3fffff60:  7bcad000 4277ecf9 7bcad000 402011b0  
3fffff70:  5a313230 00000000 000052a4 4020b1ad  
3fffff80:  3ffefd8c 0018001f 80000000 4010020c  
3fffff90:  61ffe54f 00000000 3ffeeb78 4010022d  
3fffffa0:  3fffdad0 00000000 3ffeeb78 402034c4  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

gmag11, 2022-02-13:
I guess you're trying to write too fast to serial port. But it is just a guess. Try to use exception decoder in Platformio to find the point where code crashed.

Addictek, 2022-03-19:
Hi, I faced the same problem and discovered that this is not related to serial port write speed.

The error is caused by the use of **delay(100)** statement in the main loop. I guess it is somehow interfering with ticker events scheduled by the library that are calling some asynchronous functions of TCP/UDP stack that don't tolerate the use of delay.

This error can be easily reproduced by changing delay(0) to delay(100) into the main loop of advancedExample.

I suggest it is worth mentionig this problem in the library documentation. As an improvement you could provide an alternative way to run the periodic tasks providing an update() method to be called explicitely inside the main loop without relying on ticker library.

gmag11, 2022-06-07:
I guess I will change Ticker by etsTimer so it does not depend on external Library.

Anyway you can call stop() to stop internal processes. Time will not be updated anymore but internal clock will keep on running.

There is not a single call to sync time. This is because the whole synchronization is a multiple step process that happens asynchronously.

If you don't need high precision this library provides I recommend using ESP8266 SDK internal SNTP functions.